olicooper/esphome-nspanel-lovelace-native #1: Entities missing either support or config validation

CultusMechanicus, 2024-04-16:
I've just given this a quick test by setting up a new yaml in esphome and building it (but not actually flashing it yet). Most of it transferred over pretty well, except for a few entities.

Input_button, input_boolean and covers all get rejected with :

entity_id "input_boolean.night_mode" must match the format "[entity type].[entity name]" and contain only numbers (0-9), letters (A-Z) and underscores (_), e.g. "light.living_room_light_1"

I had a look through the code and can see at least some handling for the input_* domains, but it seems either mine have odd names or the config validation is broken somehow.

olicooper, 2024-04-16:
Regex is validating the entity id's but `input_boolean` is not currently in the list of currently supported entity types. The problem is that the validation error message is not very helpful... I will update the code now to support a few extra entities and improve the error message

closed 2024-04-16 by olicooper

CultusMechanicus, 2024-04-16:
Jesus, that was quick! Thanks, I'll give it a test in an hour or so!

commit 34b34ade (2024-04-16): [codegen] Improve entity type validation.

Fixes #1