olicooper/esphome-nspanel-lovelace-native #4: timercard: no start / stop button?

kiddyfurby, 2024-04-19:
from a cardGrid entry pointing to a timer.xxx entity, I can bring up popupTimer but there's no way to start / stop / pause the timer per appdaemon backend does.

olicooper, 2024-04-19:
Firstly, thanks for testing this firmware on your nspanel! This repo is in beta at the moment, which means some cards are not implemented yet - and the timer card is one of them which is why it doesn't work yet. Please refer to the [readme](https://github.com/olicooper/esphome-nspanel-lovelace-native?tab=readme-ov-file#help-needed) for a list of currently supported cards. I will try to add support soon, please check back for updates.

olicooper, 2024-04-26:
I have added support for timers, but I think Home Assistant or ESPHome is broken because I am not receiving timer attribute updates. I have opened an issue here: https://github.com/esphome/issues/issues/5742

kiddyfurby, 2024-04-27:
Thank you!!

Home assistant do not send timer updates, to display the remaining time, even in lovelace, requires some third party module (i.e.  https://github.com/rianadon/timer-bar-card) that tries to keep an in-browser timer synchronous to HA time.

The appdaemon based backend did something similar afaik.

olicooper, 2024-04-27:
I don't think it is quite the same.. I don't need live updates, only the `finishes_at` attribute state is required for the remaining time calculation which is present when you start the timer. HA doesn't send the update to the panel though.

At the moment the alarm card will immediately close because the panel isn't getting attribute updates for the timer entity (`editable`,`duration`,`remaining`,`finishes_at`), so it can't render the page. Once the esphome / home assistant issues are fixed it should work but I will try to look for another solution too.

olicooper, 2024-05-02:
The timer should now work if you want to try it? It turns out I had an incorrect if else statement which stopped the subscriptions working!

kiddyfurby, 2024-05-02:
Right away!!

kiddyfurby, 2024-05-02:
The timer screen would still close immediately.
when I trigger the timer from HA, I do get the following log line
`[18:38:54][D][nspanel_lovelace:1305]: HA update: timer.extractor state='active'`

olicooper, 2024-05-02:
Have you pulled the latest code? There should be multiple HA updates for the timer

kiddyfurby, 2024-05-02:
Start / pause / finish now works!

Can't believe I screwed that up. I did `esphome clean` but somehow was still not on the latest commit. 
Snapped the commit sha in the yaml and we're in business.

Adjusting the timer didn't work - I was able to tap on the minute, and the +/- shows, and I was able to + the minute
but the new minute value is not "saved", I believe the following is the relevant log.
my timer was set to 0:20:00 in HA to start with, using the nspanel I adjusted it to 00:10:00, but when I tap on the minute number again (to save the new time and return to timer start page), the value is not saved and the timer is started.
```
[18:59:19][D][nspanel_lovelace:397]: Screen CMD: event,buttonPress2,uuid.18,timer-start,00:10:0
[18:59:19][D][nspanel_lovelace:1232]: Call HA: timer.start
[18:59:19][D][nspanel_lovelace:1266]: HA update: timer.extractor remaining='0:20:00'
[18:59:19][D][nspanel_lovelace:1266]: HA update: timer.extractor finishes_at='2024-05-02T19:19:19+00:00'
[18:59:19][D][nspanel_lovelace:1266]: HA update: timer.extractor state='active'
[18:59:19][D][nspanel_lovelace:227]: Render HA update
```

olicooper, 2024-05-02:
Nice! No problem, I will look in to it 👍

commit 6a71629c (2024-04-26): Add initial timer support

Closes #4
commit 408b60f9 (2024-05-02): Fix timer subscription issues

Fixes #4
commit 8285a7ec (2024-05-02): Add functionality to change timer value

Closes #4
closed 2024-05-02 by olicooper