olicooper/esphome-nspanel-lovelace-native #5: Alarm Card can't use PIN code to disarm

paulbyrne-spacenk, 2024-04-23:
The alarm card works to arm and disarm the alarm, however it doesn't show the number pad when disarming and can't disarm alarms that require a code. eg an Alarmo with 'Require code for disarming' can't be disarmed.

paulbyrne-spacenk, 2024-04-23:
Oh, I see what it's doing! Not sure if this is a bug here or with Alarmo actually. Nspanel checks  code_arm_required, which is only set to true in Alarmo if you require arming codes, not just disarming codes. Either way, I can work with that. Closing this one.

closed 2024-04-23 by paulbyrne-spacenk

CultusMechanicus, 2024-04-28:
For anyone who has this issue and is using the Alarmo integration, here's how to fix it:-
Under 'Codes' in Alarmo  there's three toggles, called "Require code for 'x'". Turn them all on.
Make sure all your user codes are set to both "Allow for Arm" and "Allow for Disarm".
The nspanel will now show the keypad where you need it.