olicooper/esphome-nspanel-lovelace-native #8: Weather on screensaver tries to read the 'forecast' attribute on weather entities, which has been removed from HA recently.

CultusMechanicus, 2024-04-23:
Not really an issue, but you need to point the screensaver weather at a template sensor like the one below as HA 2024.04 removed the 'forecast' attribute out to a service call for some random reason...
```
template:
  - trigger:
      - platform: time_pattern
        hours: /1
      - platform: homeassistant
        event: start
    action:
      - service: weather.get_forecasts
        data:
          type: daily
        target:
          entity_id: weather.home # change to your weather entity
        response_variable: daily
    sensor:
      - name: Weather Forecast Daily
        unique_id: weather_forecast_daily
        state: "{{ states('weather.home') }}" # # change to your weather entity in this line
        attributes:
          temperature: "{{ state_attr('weather.home', 'temperature') }}" # change to your weather entity
          temperature_unit: "{{ state_attr('weather.home', 'temperature_unit') }}" # change to your weather entity
          forecast: "{{ daily['weather.home'].forecast }}" # change to your weather entity
```

olicooper, 2024-04-25:
Interesting! This change shouldn't be too hard to make, but I will lose the automatic status updates so I might have to poll the service instead now. I am on HA v4.3 so that explains why I haven't spotted this yet. Thankfully the weather service is already included in v4.3.

CultusMechanicus, 2024-04-26:
Yeah, I don't quite get the reasoning they had when they moved it from an attribute on weather entities to a service call, it broke a lot of things.

olicooper, 2024-04-27:
Unfortunately esphome does not support service calls with response data so I have submitted a feature request https://github.com/esphome/feature-requests/issues/2703

andreas-bulling, 2025-01-05:
I have tried this solution but encountered a problem. The template sensor works in principle but it mixes up days and forecasts: Today's temperature and forecast are shown on the far left, slightly enlarged. As it should be.

Right next to it, however, it shows the previous day (Saturday, as I am writing this on a Sunday) but with the forecast for Monday. And so on. So the labels are shifted by two days.

[image]

olicooper, 2025-01-06:
The screen just reads the JSON data that is sent to it, so it is likely that your weather provider has the previous days weather as the first element in the array.. Have a look at the sensor state in developer tools on Home Assistant to confirm this. I think I have seen similar with my setup but after midday it refreshes to show the next day. The first (larger) icon is the 'current' weather taken from the main weather `state` and `temperature` attribute.

There might be a bug too so I will double check this but you should be able to correlate the sensor data with what you see on the screen. I would imagine you will be able to adjust what data is provided by adjusting the config in HA.

andreas-bulling, 2025-01-06:
I will try once the display works again. It was working fine yesterday but after switching it off over night (given it doesn't have an auto-off function out of the box) it doesn't start up anymore today :(

andreas-bulling, 2025-01-06:
This is what the template sensor `weather_forecast_daily` contains:

```
temperature: 10.4
temperature_unit: °C
forecast:
  - datetime: "2025-01-06T00:00:00Z"
    cloud_coverage: 96
    condition: rainy
    evaporation: null
    fog_probability: null
    precipitation_duration: 7500
    precipitation_probability: 53
    pressure: 1002.7
    sun_duration: 360
    sun_irradiance: 30
    uv_index: null
    visibility: 57200
    wind_gust_speed: 35.2
    wind_bearing: SW
    temperature: 12
    templow: 7
    dew_point: 4
    wind_speed: 16.7
    precipitation: 0.9
  - datetime: "2025-01-07T00:00:00Z"
    cloud_coverage: 96
    condition: cloudy
    evaporation: null
    fog_probability: 1
    precipitation_duration: 6660
    precipitation_probability: 32
    pressure: 1008.9
    sun_duration: 8100
    sun_irradiance: 2400
    uv_index: null
    visibility: 53400
    wind_gust_speed: 44.5
    wind_bearing: SW
    temperature: 7
    templow: 3
    dew_point: 3
    wind_speed: 22.2
    precipitation: 0
  - datetime: "2025-01-08T00:00:00Z"
    cloud_coverage: 100
    condition: rainy
    evaporation: 0.5
    fog_probability: 5
    precipitation_duration: 53400
    precipitation_probability: 77
    pressure: 1008.8
    sun_duration: 60
    sun_irradiance: 1070
    uv_index: null
    visibility: 6200
    wind_gust_speed: 24.1
    wind_bearing: S
    temperature: 6
    templow: 1
    dew_point: 5
    wind_speed: 11.1
    precipitation: 13
  - datetime: "2025-01-09T00:00:00Z"
    cloud_coverage: 100
    condition: snowy-rainy
    evaporation: 0.1
    fog_probability: 5
    precipitation_duration: 32520
    precipitation_probability: 70
    pressure: 1015.6
    sun_duration: 480
    sun_irradiance: 1160
    uv_index: null
    visibility: 7500
    wind_gust_speed: 40.8
    wind_bearing: SW
    temperature: 10
    templow: 2
    dew_point: 5
    wind_speed: 20.4
    precipitation: 7.9
  - datetime: "2025-01-10T00:00:00Z"
    cloud_coverage: 92
    condition: cloudy
    evaporation: 0.5
    fog_probability: 2
    precipitation_duration: 10740
    precipitation_probability: 52
    pressure: 1022.3
    sun_duration: 6180
    sun_irradiance: 2750
    uv_index: null
    visibility: 22600
    wind_gust_speed: 27.8
    wind_bearing: SW
    temperature: 3
    templow: -1
    dew_point: -1
    wind_speed: 13
    precipitation: 0.1
  - datetime: "2025-01-11T00:00:00Z"
    cloud_coverage: 79
    condition: cloudy
    evaporation: null
    fog_probability: 4
    precipitation_duration: 8760
    precipitation_probability: 27
    pressure: 1030.6
    sun_duration: 8160
    sun_irradiance: 2630
    uv_index: null
    visibility: 13500
    wind_gust_speed: 20.4
    wind_bearing: W
    temperature: 2
    templow: -2
    dew_point: -3
    wind_speed: 11.1
    precipitation: 0
  - datetime: "2025-01-12T00:00:00Z"
    cloud_coverage: 75
    condition: cloudy
    evaporation: null
    fog_probability: 5
    precipitation_duration: 13980
    precipitation_probability: 36
    pressure: 1037.1
    sun_duration: 6720
    sun_irradiance: 2790
    uv_index: null
    visibility: 12100
    wind_gust_speed: 18.5
    wind_bearing: SW
    temperature: 1
    templow: -2
    dew_point: -4
    wind_speed: 11.1
    precipitation: 0
  - datetime: "2025-01-13T00:00:00Z"
    cloud_coverage: 69
    condition: cloudy
    evaporation: null
    fog_probability: 6
    precipitation_duration: 8400
    precipitation_probability: 27
    pressure: 1037.2
    sun_duration: 12960
    sun_irradiance: 3070
    uv_index: null
    visibility: 10400
    wind_gust_speed: 16.7
    wind_bearing: O
    temperature: 2
    templow: -4
    dew_point: -3
    wind_speed: 9.3
    precipitation: 0
  - datetime: "2025-01-14T00:00:00Z"
    cloud_coverage: 73
    condition: cloudy
    evaporation: null
    fog_probability: 6
    precipitation_duration: 7440
    precipitation_probability: 19
    pressure: 1033.7
    sun_duration: 11880
    sun_irradiance: 3180
    uv_index: null
    visibility: 9000
    wind_gust_speed: 14.8
    wind_bearing: O
    temperature: 2
    templow: -3
    dew_point: -2
    wind_speed: 9.3
    precipitation: 0
friendly_name: Weather Forecast Daily

```

olicooper, 2025-07-24:
Hi, I realized I didn't address your second problem of the labels being shifted 2 days along. I have tested the code that is responsible for working out which weekday label to show and it seems to be working okay (see here: https://www.programiz.com/online-compiler/7CgtRwxlUp1n6) so I am not sure what was happening there (it is displaying okay on my panel?)...

I have made some updates which might be worth you trying in-case that fixes something? You'll need to update your HA weather template (see the [readme](https://github.com/olicooper/esphome-nspanel-lovelace-native/blob/dev/README.md#1-weather-forecast-is-not-displayed-on-the-screensaver-when-using-home-assistant-20244-or-later)) which will also reduce the size of data being sent to the panel.

In regards to the previous day issue, I have suggested to [other people](https://github.com/olicooper/esphome-nspanel-lovelace-native/issues/35#issuecomment-3113424780) to update HA template so that it removes the first item. It is worth noting that the weather entity updates with the next days weather as the first item after midday (for me at least) so you might not have seen the previous day entry at 6PM when you wrote the message?