olicooper/esphome-nspanel-lovelace-native #18: OTA not working

newhinton, 2024-07-29:
I was trying out this great firmware, but after first omitting the `ota:` part (due to using an older esphome), i soft bricked my device. However, regardless on how i configure ota, i cannot update my firmware again.

This is the error esphome shows:
```

INFO Uploading /config/.esphome/./.build/nspanel-basic/.pioenvs/nspanel/firmware.bin (1056976 bytes)
ERROR Error binary size: Unknown error from ESP
```

and this shows up on the device itself:
```

17:35:52	[D]	[esphome.ota:117]	Starting update from 192.168.178.104...
17:35:52	[E]	[component:164]	Component esphome.ota set Error flag: unspecified
17:35:57	[E]	[component:176]	Component esphome.ota cleared Error flag

```

I have enabled the webserver component, and preferably i'd like to use it's ability to upload firmware. Do you know how to solve this? Thanks!

newhinton, 2024-07-29:
Ah, i think the issue is that i came from tasmota.

I uploaded the .bin locally with 

`esptool.py --chip esp32 -p /dev/ttyUSB0 write_flash 0x0 nspanel.factory.bin`
and it started working.

Though i still dont have the webserver-ota.

For reference: https://github.com/esphome/issues/issues/3822

closed 2024-07-29 by newhinton

olicooper, 2024-07-30:
Yes, this is might be because tasmota and esphome partitions the flash differently? I'm really not sure though. This issue would be something to raise with esphome or tasmota repositories.

The standard esphome generated `partitions.csv` file looks like this (note the app0 and app1 partitions):
```csv
otadata,  data, ota,     ,        0x2000,
phy_init, data, phy,     ,        0x1000,
app0,     app,  ota_0,   ,        0x1C0000,
app1,     app,  ota_1,   ,        0x1C0000,
nvs,      data, nvs,     ,        0x6D000,
```

newhinton, 2024-07-31:
> Also, you should update your esphome version

That's why i initially soft-bricked my device. My older esphome version did not yet support the new ota syntax, and i just removed it xD

Afterwards it worked fine with the new version, at least the wireless flashing via esphome itself. On other non esp32 devices i get a button to directly upload firmware, but that is still missing. Tough this is a "workaround" i can totally live with.