olicooper/esphome-nspanel-lovelace-native #20: Can't update the Nextion / Display Firmware using the 'id(nspanel).upload_tft(url);' service

vmarquar, 2024-08-01:
I just bought a brand new Sonoff NS Panel and tried to flash the nspanel.tft firmware (version 53) onto it. Unfortunately, I wasn't able to complete the display firmware update. I tried several attemps and all updates freezed at random percentages (6%, 18%, 47%, 2% etc.). I used the firmware v53, downloaded from this [url](https://github.com/joBr99/nspanel-lovelace-ui/blob/v4.3.3/HMI/nspanel.tft), and hosted it on ha locally in the www-folder.

My workaround was to flash Tasmota onto the esp and using the `FlashNextion http://my-ha-ip:8123/local/nspanel.tft` command to successfully update the display firmware onto it. After that I was able to flash the esphome firmware onto the display as intended.

However, these little issues apart - thank you very much for this project! :) It looks really promising to me and I hope to get rid of the AppDaemon solution for my NS Panels.

olicooper, 2024-08-02:
Hi, thanks for trying out the project! I recently discovered this issue too and I have been trying to figure out what the cause is without success so far!.. I did push an update yesterday which improved it, are you using the latest code?

vmarquar, 2024-08-02:
Hi, 
I was using the latest code from the `dev` branch - but I only tried it on 31.07. So probably your recent changes weren't included while I was trying to flash the tft firmware.

vmarquar, 2024-08-02:
I tried searching for similar problems and found these issues:
- https://github.com/joBr99/nspanel-lovelace-ui/issues/862#issuecomment-1510483928
- https://github.com/sairon/esphome-nspanel-lovelace-ui/issues/26#issuecomment-1567269229

Maybe this helps!

onyxkyr, 2024-08-05:
Hello, I am facing a similar problem, for the me the upload process does not even start.
Setting the log-level of esphome to DEBUG I get the following log trying to upload the nspanel.tft :
```
[23:35:08][I][app:100]: ESPHome version 2024.7.3 compiled on Aug  5 2024, 23:31:22
[...]
[23:35:11][D][nspanel_lovelace_upload:194]: Nextion TFT upload requested
[23:35:11][D][nspanel_lovelace_upload:195]: URL: http://192.168.X.Y/nspanel.tft
[23:35:11][D][nspanel_lovelace_upload:209]: Exiting Nextion reparse mode
[23:35:11][D][nspanel_lovelace_upload:222]: Initializing HTTP client
[23:35:11][D][nspanel_lovelace_upload:223]: Free heap: 2154355
[23:35:11][D][nspanel_lovelace_upload:245]: Check if the client could connect
[23:35:11][D][nspanel_lovelace_upload:246]: Free heap: 2155479
[23:35:11][D][nspanel_lovelace_upload:255]: Check the HTTP Status Code
[23:35:11][D][nspanel_lovelace_upload:256]: Free heap: 2154971
[23:35:11][D][nspanel_lovelace_upload:264]: TFT file size: 297995 bytes
[23:35:11][D][nspanel_lovelace_upload:273]: File size check passed. Proceeding...
[23:35:11][D][nspanel_lovelace_upload:277]: Uploading Nextion
[23:35:11][D][nspanel_lovelace_upload:280]: Wake-up Nextion
[23:35:11][D][nspanel_lovelace:1130]: Sending: dimmode~100~100
[23:35:12][D][nspanel_lovelace_upload:286]: Free heap: 2158347
[23:35:12][D][nspanel_lovelace_upload:296]: Clear serial receive buffer
[23:35:12][D][nspanel_lovelace_upload:303]: Free heap: 2158347
[23:35:12][D][nspanel_lovelace_upload:305]: Send upload instruction: whmi-wris 297995,115200,1
[23:35:12][D][nspanel_lovelace:1130]: Sending: whmi-wris 297995,115200,1
[23:35:12][C][uart.idf:093]: Setting up UART 1...
[23:35:12][D][esp-idf:000]: I (21761) uart: queue free spaces: 20

[23:35:12][D][nspanel_lovelace_upload:319]: Waiting for upgrade response
[23:35:17][D][nspanel_lovelace_upload:325]: Upgrade response is [] - 0 byte(s)
[23:35:17][D][nspanel_lovelace_upload:326]: Free heap: 2154803
[23:35:17][E][nspanel_lovelace_upload:331]: Preparation for TFT upload failed 0 ""
[23:35:17][D][nspanel_lovelace_upload:332]: Close HTTP connection
[23:35:17][D][nspanel_lovelace_upload:335]: Connection closed
[23:35:17][D][nspanel_lovelace_upload:381]: Nextion TFT upload finished: NO
[23:35:17][D][nspanel_lovelace:1130]: Sending: rest
[23:35:17][E][nspanel_lovelace_upload:397]: Nextion TFT upload failed
[23:35:17][W][component:237]: Component api took a long time for an operation (5601 ms).
[23:35:17][W][component:238]: Components should block for at most 30 ms.
```

Reading the links above, I also tried to set the uart baud_rate to 9600, which makes no difference for me.
I reseated the flex cable to make sure its no connection issue, the display is turned on and shows a default screen; also it is possible to turn it off and on. I am wondering why the uart response is 0 byte(s) in size?

olicooper, 2024-08-06:
It sounds like your issue is different because of the nextion response.. Do you have the original TFT firmware on your panel? Is the panel blank/off when you start the uploading? I will push an update based on a theory I have.

onyxkyr, 2024-08-06:
I am quite sure, I bought a used NSPanel to test this, it appeared to be still with the default Sonoff firmware, I just flashed esphome with the basic configuration of this repo. The screen does not change at all. When I power on the device, a sonoff animation plays and then a default screen with time 0:00 and null temperatures etc. is displayed. The display can be toggled via esphome, but during flash it remains exactly like that.
I will check with the other tft file, the latest tft I tried was around 8MB in size.

olicooper, 2024-08-06:
I have pushed a commit for you to test :) Try deleting the `.esphome\.build` directory before rebuilding too (sometimes platformio has issues with caching old versions of stuff).

onyxkyr, 2024-08-06:
Is "clean build files" sufficient? Otherwise I need to find out where the build directory is.
If it is sufficient, I just tested it, and it makes no difference as far as I can tell.
```
[11:11:09][D][nspanel_lovelace:1157]: Sending: time~11:11
[11:11:31][D][nspanel_lovelace_upload:194]: Nextion TFT upload requested
[11:11:31][D][nspanel_lovelace_upload:195]: URL: http://nspanel.pky.eu/lui-release.tft
[11:11:31][D][nspanel_lovelace_upload:209]: Exiting Nextion reparse mode
[11:11:31][D][nspanel_lovelace_upload:222]: Initializing HTTP client
[11:11:31][D][nspanel_lovelace_upload:223]: Free heap: 2157639
[11:11:31][D][nspanel_lovelace_upload:245]: Check if the client could connect
[11:11:31][D][nspanel_lovelace_upload:246]: Free heap: 2155391
[11:11:31][D][nspanel_lovelace_upload:255]: Check the HTTP Status Code
[11:11:31][D][nspanel_lovelace_upload:256]: Free heap: 2156415
[11:11:31][D][nspanel_lovelace_upload:264]: TFT file size: 8373568 bytes
[11:11:31][D][nspanel_lovelace_upload:273]: File size check passed. Proceeding...
[11:11:31][D][nspanel_lovelace_upload:277]: Uploading Nextion
[11:11:31][D][nspanel_lovelace_upload:280]: Wake-up Nextion
[11:11:31][D][nspanel_lovelace:1130]: Sending: dimmode~100~100
[11:11:31][D][nspanel_lovelace_upload:286]: Free heap: 2158247
[11:11:31][D][nspanel_lovelace_upload:296]: Clear serial receive buffer
[11:11:31][D][nspanel_lovelace_upload:303]: Free heap: 2158247
[11:11:31][D][nspanel_lovelace_upload:305]: Send upload instruction: whmi-wris 8373568,115200,1
[11:11:31][D][nspanel_lovelace:1130]: Sending: whmi-wris 8373568,115200,1
[11:11:31][C][uart.idf:093]: Setting up UART 1...
[11:11:31][D][esp-idf:000]: I (28643) uart: queue free spaces: 20

[11:11:31][D][nspanel_lovelace_upload:319]: Waiting for upgrade response
[11:11:36][D][nspanel_lovelace_upload:325]: Upgrade response is [] - 0 byte(s)
[11:11:36][D][nspanel_lovelace_upload:326]: Free heap: 2155019
[11:11:36][E][nspanel_lovelace_upload:331]: Preparation for TFT upload failed 0 ""
[11:11:36][D][nspanel_lovelace_upload:332]: Close HTTP connection
[11:11:36][D][nspanel_lovelace_upload:335]: Connection closed
[11:11:36][D][nspanel_lovelace_upload:381]: Nextion TFT upload finished: NO
[11:11:36][D][nspanel_lovelace:1130]: Sending: rest
[11:11:36][E][nspanel_lovelace_upload:397]: Nextion TFT upload failed
[11:11:36][W][component:237]: Component api took a long time for an operation (5639 ms).
[11:11:36][W][component:238]: Components should block for at most 30 ms.
```

Here is how the screen looks all the time after the bootup animation
[image]

olicooper, 2024-08-06:
I only see ` Sending: dimmode~100~100` in the logs which means that your firmware doesn't include the latest commit..
Try changing the `refresh` property to something small on the esphome config?

```yaml
      url: https://github.com/olicooper/esphome-nspanel-lovelace-native
      ref: dev
    refresh: 60s  <--
```

olicooper, 2024-08-06:
Okay, so it is probably this issue... https://github.com/joBr99/nspanel-lovelace-ui/issues/862#issuecomment-1510483928
The baud rate needs to be set to 9600 for stock firmware if a previous firmware update failed. Interesting!

onyxkyr, 2024-08-06:
I tried again with the following config
```
esp32:
  board: esp32dev
  framework:
    type: arduino #esp-idf
    #sdkconfig_options:
      ## These are required to allow access to the PSRAM
      ## NOTE: This is temporary and will be removed in a 
      ##       future release when the build script is improved
    #  CONFIG_D0WD_PSRAM_CLK_IO: '5'
    #  CONFIG_D0WD_PSRAM_CS_IO: '18'

external_components:
  - source:
      type: git
      url: https://github.com/olicooper/esphome-nspanel-lovelace-native
      ref: arduino-panic-fix #dev
    refresh: 1s
    components: [nspanel_lovelace]

## UART for the Nextion display
uart:
  id: uart_nextion
  tx_pin: 16
  rx_pin: 17
  baud_rate: 9600
```

I tried the update command multiple times, getting the following feedback ( the currently updating part).

```
[11:54:16][D][nspanel_lovelace_upload:133]: Currently updating
[11:54:46][D][nspanel_lovelace_upload:133]: Currently updating
[11:57:23][D][nspanel_lovelace_upload:133]: Currently updating
[11:58:58][D][time:051]: Synchronized time: 2024-08-06 11:58:58
[11:58:58][D][time:051]: Synchronized time: 2024-08-06 11:58:58
[12:02:05][D][nspanel_lovelace_upload:133]: Currently updating
[12:13:58][D][time:051]: Synchronized time: 2024-08-06 12:13:58
[12:13:58][D][time:051]: Synchronized time: 2024-08-06 12:13:58
[12:28:58][D][time:051]: Synchronized time: 2024-08-06 12:28:58
[12:28:58][D][time:051]: Synchronized time: 2024-08-06 12:28:58
[12:34:11][D][nspanel_lovelace_upload:133]: Currently updating
```
I probably missed the initial log. I assume it should not take more than 30 minutes? Should I give it some more time, or try after a reboot?

olicooper, 2024-08-06:
You'll have to restart the esp to issue the update command again because the `is_updating_` flag is set to stop the code from starting an update during an existing update. I'd have to see more of the logs to know what is happening but the display will show an update bar if it is working properly so I imagine it probably isn't working still.. 

Right now I think your quickest option is to use the [NSPanel guide](https://docs.nspanel.pky.eu/stable/prepare_nspanel/) to flash Tasmota using a USB TTL like the CH340 and then flash ESPHome again after updating the TFT. The process is roughly as follows:
* Head over to https://tasmota.github.io/install/ and select `Tasmota32 Sonoff-NSPanel (english)` from the dropdown list, then follow the instructions to upload the firmware to your NSPanel
* Connect to the panel serial console and run this command to install the berry driver `Backlog UrlFetch https://raw.githubusercontent.com/joBr99/nspanel-lovelace-ui/main/tasmota/autoexec.be; SetOption151 0;Restart 1`
* Again in the serial console, upload the TFT `FlashNextion http://nspanel.pky.eu/lui-release.tft`
* Upload ESPHome using the USB TTL (not via the Tasmota webui as this doesn't work properly).

Once the custom TFT is uploaded, this issue shouldn't happen again based on what I've read online (although there are still other issues with the upload process I still need to fix).

onyxkyr, 2024-08-06:
Flashing via Tasmota does indeed work without any troubles, looking forward to test your project now!

olicooper, 2024-08-21:
I have added a commit which will hopefully address the issue you had. I haven't tested it though so I'm not sure it will work. Any testers out there that want to try this?

For anyone else reading this... If you have the **stock TFT firmware** installed on your device, please set the `baud_rate` to `9600` in the `uart` section of your esphome config before trying to use the TFT upload feature. This will trigger an upload using the old/legacy upload protocol.
It is a good idea to leave on `DEBUG` logging too so we have got more information to work with if there is an issue with the upload.
After the initial upload is complete, you are safe to set the `baud_rate` back to `115200` as the custom firmware uses this baud by default.

vmarquar, 2024-08-25:
@olicooper Thanks for your effort! I will test the latest changes in the upcoming days and will let you know if it works. Just to make sure I understood you correctly:
- I would pull the latest `dev` branch and test it using following uart-config.
- then I would start the `upload_tft` service from HA and log the output of the esp / nextion display via the uart port.

My minimal config looks like this:
```
(...)

external_components:
  - source:
      type: git
      url: https://github.com/olicooper/esphome-nspanel-lovelace-native
      ref: dev
    refresh: 3h
    components: [nspanel_lovelace]

## UART for the Nextion display
uart:
  id: uart_nextion
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200 #baud-rate 115.200 should be ok, since I'm already on Firmware v53?

logger:
  baud_rate: 115200
  level: DEBUG

api:
  encryption:
    key: !secret password
    
  services:
    ## Service to update the TFT firmware using a URL that is accessible to the nspanel
    - service: upload_tft
      variables:
        url: string # in my case with the local HA webserver: http://192.xxx.xxx.xxx:8123/local/nspanel.tft 
      then:
        - lambda: 'id(nspanel).upload_tft(url);'

(...)
```

----
Referrring to the stock firmware issue: Maybe it would be a good idea to update the `Readme.md` with a little Warning section to inform new users there, e.g.:
> [!IMPORTANT]  
> If you have the stock TFT firmware installed on your device, please set the `baud_rate` to `9600` in the uart section of your esphome config before trying to use the TFT upload feature.

olicooper, 2024-08-25:
Yes that looks okay to me.. The standard update procedure is mostly the same as it was before, so I would hope that updating the lovelace-ui firmware from an older version works as it did before. My main concern is whether it is possible to replace the stock firmware with the lovelace-ui firmware. I only have one NSPanel which is in active use so it is difficult to test these things sometimes 😄

If you look at the variable `update_baud_rate_` you'll see that it actually switches to `921600` baud to do the update. This is static right now which might be an issue based on how I've seen Tasmota handle it, but I wanted to provide a low friction way for people to update when the standard way fails (it might not fail for everyone, so sometimes it might not be necessary to change the baud to 9600).

Yes, that is a good idea to add a warning to the readme, but to avoid confusion, I think I will hold off adding that until I have got confirmation that the fix works.

vmarquar, 2024-08-26:
I've just tested the upload_tft() feature but unfortunately it didn't work. As soon as i send the upload_service() the nextion display freezes and isn't usable anymore. I collected following logs:

```
[17:33:45][D][nspanel_lovelace:1299]: TFT CMD OUT: pageType~screensaver
[17:33:45][D][nspanel_lovelace:1299]: TFT CMD OUT: timeout~20
[17:33:45][D][nspanel_lovelace:1299]: TFT CMD OUT: weatherUpdate~~~~31728~~15.6°C~~~~31728~Mo~15.7°C~~~~31728~Di~22.7°C~~~~65504~Mi~27.8°C~~~~65504~Do~28.8°C
[17:33:57][I][nspanel_lovelace_upload:198]: Nextion TFT upload requested
[17:33:57][D][nspanel_lovelace_upload:199]: URL: http://192.168.178.2:8123/local/nspanel.tft
[17:33:57][I][nspanel_lovelace_upload:217]: Using protocol v1.2
[17:33:57][D][nspanel_lovelace_upload:219]: Exiting Nextion reparse mode
[17:33:57][D][nspanel_lovelace_upload:232]: Initializing HTTP client
[17:33:57][D][nspanel_lovelace_upload:233]: Free heap: 2210727
[17:33:57][D][nspanel_lovelace_upload:255]: Check if the client could connect
[17:33:57][D][nspanel_lovelace_upload:256]: Free heap: 2213147
[17:33:57][D][nspanel_lovelace_upload:265]: Check the HTTP Status Code
[17:33:57][D][nspanel_lovelace_upload:266]: Free heap: 2212347
[17:33:57][D][nspanel_lovelace_upload:274]: TFT file size: 8371116 bytes
[17:33:57][D][nspanel_lovelace_upload:283]: File size check passed. Proceeding...
[17:33:57][D][nspanel_lovelace_upload:288]: Wake-up Nextion
[17:33:57][D][nspanel_lovelace:1272]: Sending: dimmode~100~100
[17:33:58][D][nspanel_lovelace_upload:310]: Clear serial receive buffer
[17:33:58][D][nspanel_lovelace_upload:317]: Free heap: 2212635
[17:33:58][D][nspanel_lovelace:1272]: Sending: whmi-wris 8371116,921600,1
[17:33:58][C][uart.idf:153]: UART 1 was reloaded.
[17:33:58][C][uart.idf:159]: UART Bus 1:
[17:33:58][C][uart.idf:160]:   TX Pin: GPIO16
[17:33:58][C][uart.idf:161]:   RX Pin: GPIO17
[17:33:58][C][uart.idf:163]:   RX Buffer Size: 256
[17:33:58][C][uart.idf:165]:   Baud Rate: 921600 baud
[17:33:58][C][uart.idf:166]:   Data Bits: 8
[17:33:58][C][uart.idf:167]:   Parity: NONE
[17:33:58][C][uart.idf:168]:   Stop bits: 1
[17:33:58][D][nspanel_lovelace_upload:333]: Waiting for upgrade response
[17:34:03][D][nspanel_lovelace_upload:339]: Upgrade response is [] - 0 byte(s)
[17:34:03][D][nspanel_lovelace_upload:340]: Free heap: 2212339
[17:34:03][E][nspanel_lovelace_upload:345]: Preparation for TFT upload failed 0 ""
[17:34:03][D][nspanel_lovelace_upload:346]: Close HTTP connection
[17:34:03][D][nspanel_lovelace_upload:349]: Connection closed
[17:34:03][I][nspanel_lovelace_upload:395]: Nextion TFT upload finished: NO
[17:34:03][D][nspanel_lovelace:1272]: Sending: rest
[17:34:03][E][nspanel_lovelace_upload:411]: Nextion TFT upload failed
[17:34:03][W][component:237]: Component api took a long time for an operation (5635 ms).
[17:34:03][W][component:238]: Components should block for at most 30 ms.
[17:34:03][D][nspanel_lovelace:1299]: TFT CMD OUT: time~17:34
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 00000000
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 0080
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 0080
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 0080
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 0080
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 00000080
[17:34:58][W][nspanel_lovelace:347]: Unparsed data: 80
(...)
```

olicooper, 2024-08-27:
I will have to look in to that, I am hoping that a recent update to the TFT communication mechanism hasn't messed this up. Thanks for testing 👍

olicooper, 2024-08-28:
I've added another update which fixes some of the issues found during my testing today. It seems to work for my panel even when restarting a failed upload.

Originally I was doing all my testing on a random ESP32 test device (which is way easier to test on) but this only simulates the TFT responses which is why I didn't spot the issues you encountered 🤦 

It is not required to set 9600 baud now as I loop through some common baud rates to figure it out instead (so no need for the readme update!). I haven't tested the upload from a stock FW to the custom FW to confirm this works though.

ditheo, 2024-09-13:
Well I have the same proble. I tried to update the tft without success Here is my Log:
`INFO Successfully connected to nspanel @ 10.100.0.153 in 7.187s
INFO Successful handshake with nspanel @ 10.100.0.153 in 0.031s
[17:21:04][D][nspanel_lovelace:1290]: TFT CMD OUT: time~17:21
[17:21:04][D][nspanel_lovelace:1290]: TFT CMD OUT: date~Friday, 13. September 2024
[17:21:04][D][nspanel_lovelace:1290]: TFT CMD OUT: time~17:21
[17:21:04][D][nspanel_lovelace:2121]: HA update: light.living_room supported_color_modes='onoff'
[17:21:04][D][nspanel_lovelace:2121]: HA update: light.living_room color_mode=''
[17:21:04][D][nspanel_lovelace:2121]: HA update: light.living_room state='off'
[17:21:04][D][nspanel_lovelace:2121]: HA update: light.dining_room supported_color_modes='onoff'
[17:21:04][D][nspanel_lovelace:2121]: HA update: light.dining_room color_mode=''
[17:21:04][D][nspanel_lovelace:2121]: HA update: light.dining_room state='off'
[17:21:04][D][nspanel_lovelace:2121]: HA update: light.kitchen_2 supported_color_modes='onoff'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.kitchen_2 color_mode=''
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.kitchen_2 state='off'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.sink supported_color_modes='onoff'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.sink color_mode=''
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.sink state='off'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.souzanna_bedroom supported_color_modes='onoff'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.souzanna_bedroom color_mode=''
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.souzanna_bedroom state='off'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.chronis_bedroom supported_color_modes='onoff'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.chronis_bedroom color_mode=''
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.chronis_bedroom state='off'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.bedroom_light supported_color_modes='onoff'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.bedroom_light color_mode=''
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.bedroom_light state='off'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.bathroom supported_color_modes='onoff'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.bathroom color_mode=''
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.bathroom state='off'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.bathroom_small_right supported_color_modes='onoff'
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.bathroom_small_right color_mode=''
[17:21:05][D][nspanel_lovelace:2121]: HA update: light.bathroom_small_right state='off'
[17:21:05][D][nspanel_lovelace:2121]: HA update: sensor.boiler_outdoortemp device_class='temperature'
[17:21:05][D][nspanel_lovelace:2121]: HA update: sensor.boiler_outdoortemp unit_of_measurement='°C'
[17:21:05][D][nspanel_lovelace:2121]: HA update: sensor.boiler_outdoortemp state='28.2'
[17:21:05][D][nspanel_lovelace:2121]: HA update: sensor.temperature device_class='temperature'
[17:21:05][D][nspanel_lovelace:2121]: HA update: sensor.temperature unit_of_measurement='°C'
[17:21:05][D][nspanel_lovelace:2121]: HA update: sensor.temperature state='31.5'
[17:21:05][D][nspanel_lovelace:2121]: HA update: sensor.kitchen_temperature device_class='temperature'
[17:21:05][D][nspanel_lovelace:2121]: HA update: sensor.kitchen_temperature unit_of_measurement='°C'
[17:21:05][D][nspanel_lovelace:2121]: HA update: sensor.kitchen_temperature state='27.43'
[17:21:05][D][nspanel_lovelace:2121]: HA update: alarm_control_panel.alarmo code_arm_required='on'
[17:21:05][D][nspanel_lovelace:2121]: HA update: alarm_control_panel.alarmo open_sensors=''
[17:21:05][D][nspanel_lovelace:2121]: HA update: alarm_control_panel.alarmo state='disarmed'
[17:21:05][D][nspanel_lovelace:2121]: HA update: climate.thermostat_hc1 temperature='20.0'
[17:21:05][D][nspanel_lovelace:2121]: HA update: climate.thermostat_hc1 current_temperature='27.3'
[17:21:05][D][nspanel_lovelace:2121]: HA update: climate.thermostat_hc1 target_temp_step='0.5'
[17:21:05][D][nspanel_lovelace:2121]: HA update: climate.thermostat_hc1 min_temp='5.0'
[17:21:05][D][nspanel_lovelace:2121]: HA update: climate.thermostat_hc1 max_temp='30.0'
[17:21:05][D][nspanel_lovelace:2121]: HA update: climate.thermostat_hc1 hvac_modes='auto,heat,off'
[17:21:05][D][nspanel_lovelace:2121]: HA update: climate.thermostat_hc1 state='auto'
[17:21:49][D][sensor:094]: 'ntc_source': Sending state 1.33600 V with 2 decimals of accuracy
[17:21:49][D][resistance:039]: 'resistance_sensor' - Resistance 7618.7Ω
[17:21:49][D][sensor:094]: 'resistance_sensor': Sending state 7618.73730 Ω with 1 decimals of accuracy
[17:21:49][D][ntc:026]: 'Temperature' - Temperature: 31.2°C
[17:21:49][D][sensor:094]: 'Temperature': Sending state 31.24897 °C with 1 decimals of accuracy
[17:21:49][D][nspanel_lovelace:2121]: HA update: sensor.temperature state='31.2'
[17:21:53][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[17:21:53][D][esp32.preferences:114]: Saving 1 preferences to flash...
[17:21:53][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[17:22:00][D][nspanel_lovelace:1290]: TFT CMD OUT: time~17:22
[17:22:49][D][sensor:094]: 'ntc_source': Sending state 1.34100 V with 2 decimals of accuracy
[... cut ...]
          - switch.turn_off: screen_power
          - delay: 0.5s
          - switch.turn_on: screen_power
      ####### Single long click - toggle light relay 2
      - timing:
          - ON for 1.2s to 4s
          - OFF for at least 0.4s
        then:
          - switch.toggle: relay_2

button:
  - platform: restart
    id: restart_switch
    name: Restart

output:
  ## Buzzer for playing tones
  - platform: ledc
    id: buzzer_out
    pin: 21

## Rtttl function for buzzer
rtttl:
  id: buzzer
  output: buzzer_out`

olicooper, 2024-09-14:
It actually looks like your upload worked fine...

RAM usage fine, TFT reports a new range and continues uploading from 85.36%:
> [17:24:55][D][nspanel_lovelace_upload:144]: Uploaded 0.05%, remaining 8369472 bytes, free heap: 98896 (DRAM) + 2090519 (PSRAM) bytes, lblk 2064384
> [17:24:55][D][nspanel_lovelace_upload:155]: recv_string [08.00.00.6D.00 (5)]
> [17:24:55][I][nspanel_lovelace_upload:161]: Nextion reported new range 7143424
> [17:24:57][D][nspanel_lovelace_upload:144]: Uploaded 85.36%, remaining 1226048 bytes, free heap: 97012 (DRAM) + 2090519 (PSRAM) bytes, lblk 2064384

Then the upload finishes and the panel reboots:
> [17:26:54][D][nspanel_lovelace_upload:144]: Uploaded 100.00%, remaining 0 bytes, free heap: 96484 (DRAM) + 2090519 (PSRAM) bytes, lblk 2064384
> [17:26:54][I][nspanel_lovelace_upload:432]: Nextion TFT upload finished

What problems are you seeing?

ditheo, 2024-09-14:
The odd thing that the nextion displays only system data error. In the log
I can see : TFT CMD time out

olicooper, 2024-09-14:
I don't think I have seen this before.. if the TFT reports that it has worked then it should be fine.
Is this the error you are seeing? https://ewelinkcommunity.net/wp-content/uploads/2023/02/NSPanel_System_Data_ERROR-768x1024.jpg
The UART bus which this library uses to communicate with the TFT doesn't output any command 'time out' logs from what I recall, where are you seeing this?

Maybe try these steps?
- Remove the power from the nspanel to fully reboot the whole nspanel
- Upload an older version of the TFT firmware (maybe the file is corrupt)
- If that doesn't work, use a different server to upload (maybe the server is causing problems) e.g. http://nspanel.pky.eu/lui-release.tft

ditheo, 2024-09-14:
[17:21:04][D][nspanel_lovelace:1290]: TFT CMD OUT: time~17:21

It is on log 3rd and 4th row , also after the upload I can see that.  If
you see on the log I have no information about the nextion display such as
firmware serial etc

olicooper, 2024-09-14:
Actually, hold off on that for a moment, I am going to update the code to add a delay before restarting the panel to see if that helps.

ditheo, 2024-09-14:
[Remove the power from the nspanel to fully reboot the whole nspanel
Upload an older version of the TFT firmware (maybe the file is corrupt)
If that doesn't work, use a different server to upload (maybe the server is
causing problems) e.g. http://nspanel.pky.eu/lui-release.tft]

I have done these steps , also I tried the blank tft without success

olicooper, 2024-09-14:
Try uploading my latest changes to your panel and try the TFT upload again after a full reboot?

olicooper, 2024-09-14:
> It is on log 3rd and 4th row , also after the upload I can see that.  If
you see on the log I have no information about the nextion display such as
firmware serial etc

These logs are normal, they just tell you that the system has sent a command 'OUT' to the TFT. When the system receives a command it will say `TFT CMD IN` too (when pressing buttons on the screen etc.)
The system will not output firmware or serial numbers as this hasn't been implemented. It requires sending specific commands to the display to get this info but it isn't required the the system to function.

ditheo, 2024-09-14:
sorry , I didn't understand that, when I saw the log. Is it possible to write the commands that can give informations about the nextion display?

ditheo, 2024-09-14:
> Try uploading my latest changes to your panel and try the TFT upload again after a full reboot? (make sure esphome has actually pulled the latest changes and is not using a cached older version)

I am not right now on home. I have clean build files and I will try to cut off power to reboot and then try the new code

ditheo, 2024-09-15:
Tried again,  after a reboot ( I cut the power to nspanel for 2 minutes) and it didn't work . Maybe something is broken from the last nspanel.tft that was used. I tryto upload the panel.tft via direct serial or I try with tasmota and back to esphome

olicooper, 2024-09-15:
Just to confirm, can you see [these updates](https://github.com/olicooper/esphome-nspanel-lovelace-native/blob/76e354865bf856e6fd2706e30b54eb050cf04c17/components/nspanel_lovelace/nspanel_lovelace_upload_idf.cpp#L437-L438) in the `nspanel_lovelace_upload_idf.cpp` file on your PC? I want to know if esphome has downloaded the latest changes. You could also try `esphome clean config.yaml` before building.

It is probably better to try Tasmota first as we know this works. I really don't know what is wrong with the TFT, maybe it could be faulty? We will know after uploading with Tasmota I think.

ditheo, 2024-09-22:
I tasmotized the device , then I uplaod the tft via tasmotaand then back to your esphome yaml , the device works again.
For your reference all firmware updates are done via uart and not OTA!!! Keep in mind that tasmota after v13 use different partitions and on ESP32 it is complicated to do via OTA migrations! Use serial it is much safer!!!!

marcelveldt, 2024-10-17:
Alternative workaround for now is using this to get the TFT file uploaded to the device:
https://github.com/sairon/esphome-nspanel-lovelace-ui

And then switch to this firmware.

olicooper, 2024-10-17:
@marcelveldt Has this method worked for you? The upload process is similar between these libraries so it shouldn't make a difference in theory. Tasmota is able to upgrade from the original NSPanel firmware which is what my recent updates tried to address (these aren't accounted for in the sairon's library) but there is probably something I am still missing to make it work reliably. PR's are welcome if anyone wants to figure it out👍

marcelveldt, 2024-10-17:
> @marcelveldt Has this method worked for you? The upload process is similar between these libraries so it shouldn't make a difference in theory. Tasmota is able to upgrade from the original NSPanel [...]

Yes, I just flashed a device that way. Worked perfectly fine. After that switched back to your version.

olicooper, 2024-10-17:
Interesting.. do you have any logs? Maybe the code updates made it worse for some use cases?

andreas-bulling, 2025-01-04:
Hm, but that other project requires the `arduino` framework while yours supports `esp-idf` (which I'd prefer). But they both have different boot partitions and require serial flashing. Is this what you did @marcelveldt ?

andreas-bulling, 2025-01-04:
Great - I tried to flash the device using your firmware but it failed. Then I found this issue and understood why. But now the only thing I see is "System Data ERROR" upon startup. I reflashed it with the firmware @marcelveldt suggested but the device still doesn't connect to HA. So I cannot upload the TFT.

What shall I do now?

andreas-bulling, 2025-01-05:
After hours of trying to get this to work, I finally flashed the NSPanel_HA_Blueprint ESP firmware (https://github.com/Blackymas/NSPanel_HA_Blueprint) and, with it, `lui-release.tft`. Then I flashed the esphome-nspanel-lovelace firmware over it and all is finally working.