olicooper/esphome-nspanel-lovelace-native #21: Feature request: support for more entities like: number, input_number

Chreece, 2024-08-03:
Firstly I just want to thank you for this one-man wonderlful project.
Do you plan supporting more entities like: `number`, `input_number` and eventually something tricky like `input_text`?

olicooper, 2024-08-06:
I'd like to get the project as close as I can to the upstream nspanel project (though some features are not possible with esphome right now). If [nspanel-lovelace-ui](https://github.com/joBr99/nspanel-lovelace-ui) supports it then I should be able to add support in this project too 👍

It looks like input_text is available on the [entities card](https://docs.nspanel.pky.eu/stable/card-entities/) but read-only.. Adding a full multi-lingual keyboard probably wont work for the nspanel though. You'd have to file a feature request  on the upstream repo for that 😃

Chreece, 2024-08-07:
The entities I mentioned except the `input_text`, are already supported by nspanel-lovelace-ui.
The reason I asked for the support is because I'm using them in my current setup and wanted to switch to yours....

olicooper, 2024-08-19:
This should provide support for the entities you mentioned.
It took a lot longer than I expected because I ran in to memory consumption issues which surfaced after I added the new entity types. A lot has changed so hopefully it works still!

closed 2024-08-19 by olicooper

Chreece, 2024-08-27:
Wanted to try it by updating via OTA (was already using esphome with the https://github.com/sairon/esphome-nspanel-lovelace-ui backend). There seems to be no communication with homeassistant. 
Before investigating further, is it necessary to flash your version only via Serial for the first time?

olicooper, 2024-08-27:
In theory it should just work.. but it sounds like the upload failed or your settings were different? Is the device `name` and the `api -> encryption -> key` the same as your old config? If you added `ota` to the config and it was carshing, then after 10 attempts esphome will put it in safe-mode so you can do ota again.

Chreece, 2024-08-28:
The name is the same, also tried to remove the old esphome device and reinstall it with your config.
I haven't any encryption key used.
There seems that it gets the time and some entity names but can't control anything and the states are uknown - wrong.
The entity names aren't all in english, could that be a problem?

olicooper, 2024-08-28:
You are probably correct that it isn't connected to HomeAssistant.
The time will be sent from your PC when you connect to it to view the logs (so it must have a wifi connection).
The entity names might be taken from your config when you set it up - especially if they are not in English.. but this shouldn't cause any issues.

Are you able to get the logs from the device so I can understand the problem more? It would be better to connect to the panel directly as some logs (including any crash logs) are missed when connecting remotely over wifi.

Chreece, 2024-09-07:
I'm surely missing something...
How does it communicate with HA? Do i still need the Appdaemon Backend?
I don't see any line that directs to HA in the yaml configuration

olicooper, 2024-09-07:
No problem, I can explain...
You don't need AppDaemon (I started the project because I didn't want to have AppDaemon running too).

This project uses esphome to communicate directly with HA using an 'add-on' called 'esphome' (see here: https://esphome.io/guides/getting_started_hassio). Once you have added the 'esphome' add-on to HA your nspanel should be automatically discovered by HA (you should get a notification of a new device found). If HA doesn't find the nspanel then you may need to add it manually, but the wizard should guide you through configuration anyway. Just make sure the panel and HA can communicate with each other (they are on the same LAN subnet and the firewall on the HA machine is configured correctly etc).

Chreece, 2024-09-08:
So it really needs the addon, the esphome integration isn't enough?

olicooper, 2024-09-08:
I don't use the esphome dashboard (I upload from my PC manually)... Having read the [esphome dashboard](https://esphome.io/guides/getting_started_command_line.html) and [esphome HA add-on](https://esphome.io/guides/getting_started_hassio) docs it sounds like both do similar things so I am not sure. I may have to test it and come back to you. [This article](https://peyanski.com/complete-esphome-installation-guide/) mentions 4 different ways of using esphome!

My understanding was that the esphome dashboard/integration is only for uploading firmware and managing the device (logs etc.) but doesn't communicate with HA at all. I might be wrong though.
The HA add-on provides a link between HA and all the esphome devices.

Chreece, 2024-09-09:
Thank you very much for your time!
I'm getting seeing this message on the logs (wireless through esphome Dashboard):
```
[18:15:35][D][api.connection:1389]: Home Assistant 2024.9.1 (192.168.178.57): Connected successfully
```
So it should communicate with esphome since I added it in esphome integration. Also the switch entities are shown but trying to control them, makes them unavailable for a second and then stay again on.

After a while it just stops communicating with the dashboard and the HA and doesn't change pages when i touch it (stays on screensaver)

Sorry if I'm getting you in trouble

santibur06, 2024-09-09:
HI @Chreece. The ESPHome dashboard is just a tool to build and upload your configurations, it has nothing to do with esphome's communication with HA, in fact, you can perfectly use esphome with HA without the esphome dashboard installed. ESPHome uses HA's api to connect and communicate. At boot, esphome will 'announce' itself in the network, HA will pick that up and show it to you in the Integrations page as a newly discovered device, if you have autodiscovery enabled (which is HA's default). If it doesn't show up as discovered, then you can manually add it using the device's IP address.
Now, if you are having communication issues between HA and ESPHome (maybe you see some reconnection attempts in the device's logs?) then you may wanna take a look at your network configuration, you may have some issues there. Again, the dashboard has nothing to do with how ESPHome communicates with HA's API which is what this project uses to communicate with HA

olicooper, 2024-09-09:
@santibur06 Thanks for the detailed explanation, I wasn't able to comment on the dashboard as I have never used it!

> After a while it just stops communicating with the dashboard and the HA and doesn't change pages when I touch it (stays on screensaver)

Unfortunately this sounds like a bug that is making the device crash... can I see the section of your config where you are using this switch including the page type? Hopefully I can reproduce it, but logs would help a lot too!

Do the other entities work okay though? I don't have access to test all sensors and switches on HA so it makes testing difficult 🤦

Chreece, 2024-09-09:
@santibur06 thank you very much! My thoughts was also that Dashboard isn't in fact needed. 

@olicooper here is my configuration:

```
esphome:
  name: nspanel-wohnzimmer
  name_add_mac_suffix: true
  comment: "ESPHome Advanced NSPanel Example"
  build_path: ./.build/nspanel-advanced

#bluetooth_proxy:

#esp32_ble_tracker:
#  id: ble_tracker
  
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      ## These are required to allow access to the PSRAM
      ## NOTE: This is temporary and will be removed in a 
      ##       future release when the build script is improved
      CONFIG_D0WD_PSRAM_CLK_IO: '5'
      CONFIG_D0WD_PSRAM_CS_IO: '18'

external_components:
  - source:
      type: git
      url: https://github.com/olicooper/esphome-nspanel-lovelace-native
      ref: dev
    refresh: 3h
    components: [nspanel_lovelace]

nspanel_lovelace:
  id: nspanel
  sleep_timeout: 30
  # locale:
    # temperature_unit: celcius
    # day_of_week_map:
    #   sunday: ['Sun', 'Sunday']
    #   monday: ['Mon', 'Monday']
    #   tuesday: ['Tue', 'Tuesday']
    #   wednesday: ['Wed', 'Wednesday']
    #   thursday: ['Thu', 'Thursday']
    #   friday: ['Fri', 'Friday']
    #   saturday: ['Sat', 'Saturday']
  screensaver:
    time_id: homeassistant_time
    ## For formatting options see: https://cplusplus.com/reference/ctime/strftime/
    # time_format: '%H:%M'
    # date_format: '%A, %d. %B %Y'
    weather:
      entity_id: sensor.weather_forecast_hourly
      ## NOTE: use this version (after following the instructions on the README) with Home Assistant v4.4.0+
      # entity_id: sensor.weather_forecast_daily
    # status_icon_left:
    #   entity_id: light.front_room_inner
    #   icon: hiking
    # status_icon_right:
    #   entity_id: light.front_room_outer
    #   icon:
    #     value: robot-outline
    #     color: 200
  cards:
    - type: cardEntities
      title: Φωτισμός-Ρολά
      sleep_timeout: 10
      entities:
[... 255 lines cut ...]
          - delay: 0.5s
          - switch.turn_on: screen_power
      ####### Single long click - toggle light relay 2
      - timing:
          - ON for 1.2s to 4s
          - OFF for at least 0.4s
        then:
          - switch.toggle: relay_2

button:
  - platform: restart
    id: restart_switch
    name: Restart

output:
  ## Buzzer for playing tones
  - platform: ledc
    id: buzzer_out
    pin: 21

## Rtttl function for buzzer
rtttl:
  id: buzzer
  output: buzzer_out
```

olicooper, 2024-09-09:
I don't see any switch entities in your config (under `nspanel_lovelace > cards`).. Can you explain what you mean when your say 'switch entities'? Are you pressing the physical buttons below the screen? Does the panel still freeze even if you don't interact with it or do you have to visit a specific page for it to freeze? Everything in your config looks okay to me.
I wonder if it is running out of memory?

Chreece, 2024-09-10:
The Panel exposes the 2 physical Buttons as switches in HA.
I will try to read the serial logs when I have time

olicooper, 2024-09-10:
Ah okay.. So this part of the config is actually maintained by esphome (only the stuff under the `napanel_lovelace` section is part of this project). It is still possible that my code is causing an issue but we need to consider issues with esphome and memory exhaustion too...
- Firstly we need to start from a known point, are you using the [latest esphome](https://github.com/esphome/esphome/releases) and [HMI version 53](https://github.com/joBr99/nspanel-lovelace-ui/tree/v4.4.0/HMI)?
- Have you tried using the basic config to see if it works without freezing?
  Maybe try this config...
<details>

<summary>nspanel_test1.yaml</summary>

```yaml
esphome:
  name: nspanel-wohnzimmer
  name_add_mac_suffix: true
  comment: "NSPanel Test 1"
  build_path: ./.build/nspanel-test1

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      ## These are required to allow access to the PSRAM
      ## NOTE: This is temporary and will be removed in a 
      ##       future release when the build script is improved
      CONFIG_D0WD_PSRAM_CLK_IO: '5'
      CONFIG_D0WD_PSRAM_CS_IO: '18'

external_components:
  - source:
      type: git
      url: https://github.com/olicooper/esphome-nspanel-lovelace-native
      ref: dev
    refresh: 3h
    components: [nspanel_lovelace]

nspanel_lovelace:
  id: nspanel
  sleep_timeout: 30
  screensaver:
    time_id: homeassistant_time
    weather:
      entity_id: sensor.weather_forecast_daily
  cards:
    - type: cardEntities
      title: Φωτισμός-Ρολά
      sleep_timeout: 10
      entities:
        - entity_id: navigate.lights
          icon: lightbulb-group
        - entity_id: cover.wohnzimmer_cover_links_cover_3
        - entity_id: cover.wohnzimmer_cover_rechts_cover_6
    - type: cardGrid
      id: lights
      title: Φωτισμός
      sleep_timeout: 5
      hidden: true
      entities:
        - entity_id: light.wohnzimmer_licht_1
        - entity_id: light.wohnzimmer_licht_2
        - entity_id: light.wohnzimmer_licht_3
        - entity_id: light.wohnzimmer_licht_4
        - entity_id: light.wohnzimmer_licht_5
        - entity_id: light.wohnzimmer_licht_6

logger:
  # If the baud rate is set too slow then logs will take longer to print and 
  # the system will slow down. 115200 is a good speed to use here.
  baud_rate: 115200
  level: DEBUG
  logs:
    adc: WARN
    api: DEBUG
    app: INFO
    uart: WARN
    time: WARN
    switch: DEBUG
    sensor: WARN
    button: WARN
    text_sensor: WARN
    resistance: WARN
    mdns: WARN
    ntc: WARN
    ota: INFO
    nspanel_lovelace_upload: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.178.88

api:
  # Ideally an encryption key should be used too...
  # encryption: 
  #   key: !secret api_key
  services:
    ## Service to update the TFT firmware using a URL that is accessible to the nspanel
    - service: upload_tft
      variables:
        url: string
      then:
        - lambda: 'id(nspanel).upload_tft(url);'

ota:
  platform: esphome
  password: !secret ota_update

## UART for the Nextion display
uart:
  id: uart_nextion
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200

time:
  - platform: homeassistant
    id: homeassistant_time
    timezone: Europe/London
    on_time:
      ## At midnight
      - seconds: 0
        minutes: 0
        hours: 0
        then:
          - lambda: 'id(nspanel).set_display_dim(1, 50);'
      ## At 8:00 PM
      - seconds: 0
        minutes: 0
        hours: 20
        then:
          - lambda: 'id(nspanel).set_display_dim(20, 80);'
      ## At 7:00 AM
      - seconds: 0
        minutes: 0
        hours: 7
        then:
          - lambda: 'id(nspanel).set_display_dim(50, 100);'

switch:
  ## Physical relay 1
  - platform: gpio
    name: Relay 1
    id: relay_1
    pin:
      number: 19
    restore_mode: ALWAYS_ON
  ## Physical relay 2
  - platform: gpio
    name: Relay 2
    id: relay_2
    pin:
      number: 22
    restore_mode: ALWAYS_ON
  ## Turn screen power on/off
  - platform: gpio
    name: Screen Power
    id: screen_power
    entity_category: config
    pin:
      number: 4
      inverted: true
    restore_mode: ALWAYS_ON

binary_sensor:
  ## Left button below the display
  - platform: gpio
    name: Left Button
    pin:
      number: GPIO14
      inverted: true
    id: left_button
    internal: true
    on_click:
      - switch.toggle: relay_1
  ## Right button below the display
  - platform: gpio
    name: Right Button
    pin:
      number: GPIO27
      inverted: true
    id: right_button
    internal: true
    on_click:
      - switch.toggle: relay_2
```

</details>

Chreece, 2024-09-10:
Thank you very much for your time, I will try to find some time and test it tomorrow

Chreece, 2024-09-14:
So we are getting better...

Finally can control entities from the nspanel. The names aren't shown maybe because they aren't in English (they are in Greek though)

olicooper, 2024-09-15:
It sounds like the panel is crashing to me.

Can you try using daily weather forecast instead of hourly? There are over 30 hours of forecast returned on my instance and it might be too much for the device to handle...

If that doesn't work then the only option is to plug it in to a debugger and get the crash logs - a wifi connection won't show the crash logs.

olicooper, 2024-09-15:
I've just realised that I don't subscribe to `friendly_name` updates (to reduce the load on the device) so that is why you don't see the entity names. You will have to manually set the entity names in your config.
I should probably use the entity_id as the name by default but this project is beta currently so I haven't added all the features yet and have certainly missed some things too!

Chreece, 2024-10-12:
Sorry for the late reply, just found some time to play with the firmware.
Thank you a lot! I switched to a daily weather update and everything works as intented!
Yes you are right, i put the `name:` for every entity and now I can see the entity names.

Thanks again!

Chreece, 2024-10-13:
One question... Sorry for bothering! Have you managed to use the `bluetooth_proxy:`?
It seems to stop communicating, after enabling this... It worked with the esphome from sairon. Is it too heavy together with the native component?

olicooper, 2024-10-13:
Glad to hear it is working for you now! Because of the change to the weather entity I suspect that all of these issues are memory related. Bluetooth uses **a lot** of ram so that is probably your issue. I haven't tried it myself yet (because I don't need that feature in my setup) but I have been intending to try something to improve the RAM situation which you could try and report back to me? Can you add `CONFIG_SPIRAM_USE_MALLOC` (see below) to your config? This will change how the external RAM can be used in the device which _may_ allow it to work better. Let me know if that helps 😃
```yaml
  framework:
    sdkconfig_options:
      CONFIG_SPIRAM_USE_MALLOC: 'y'
```

Chreece, 2024-10-13:
I tried adding the `CONFIG_SPIRAM_USE_MALLOC: 'y'` with `bluetooth_proxy:` but unfortunately didn't change anything in my case. As soon as I add the bt component I don't get any wireless communication to read logs (time out connecting, thus making the debug component useless) but I can still update via ota to remove this function and have a normal operation.
If you have any other suggestions, please feel free to share, happy to help since I really like your approach above any other

olicooper, 2024-10-13:
No problem, thanks for trying! I will try to have a look at `bluetooth_proxy` but I don't have much time right now so it might be a while. Could you keep `CONFIG_SPIRAM_USE_MALLOC: 'y'` in your config for a while and see if it is stable? You might be able to use hourly forecasts again with this option enabled?

commit 8539a212 (2024-08-19): Merge: Add support for more entities

Resolves #21

* Support for: button, number, input_number, text, input_text, lock, sun, person, vacuum, weather
* New popupFan entity detail page
* Add translations for remaining UI strings
* Fix cover action icons showing 'disable' text
* Fix alarm page not auto updating on state change
* Fix panic when entity type doesn't exist
* Show RAM usage when logging config
* Add popup page entity checks