olicooper/esphome-nspanel-lovelace-native #22: [Feature] Screensaver Entities and Locales

r0bb10, 2024-08-14:
Hi!

I've finally managed to test your esphome port for the nspanel, i wouls say it's pretty damn cool and works great so far (not tested extensively but i'm going to).

I've noticed some differences for now with the original tasmota/appdaemon version that maybe could be implemented:

```
nspanel_lovelace:
  id: nspanel
  sleep_timeout: 10
  locale:
    temperature_unit: celcius
    # day_of_week_map:
    #   sunday: ['Sun', 'Sunday']
    #   monday: ['Mon', 'Monday']
    #   tuesday: ['Tue', 'Tuesday']
    #   wednesday: ['Wed', 'Wednesday']
    #   thursday: ['Thu', 'Thursday']
    #   friday: ['Fri', 'Friday']
    #   saturday: ['Sat', 'Saturday']
  screensaver:
    time_id: homeassistant_time
    ## For formatting options see: https://cplusplus.com/reference/ctime/strftime/
    # time_format: '%H:%M'
    # date_format: '%A, %d. %B %Y'
    weather:
      # entity_id: weather.openweathermap
      ## NOTE: use this version (after following the instructions on the README) with Home Assistant v4.4.0+
      entity_id: sensor.weather_forecast_daily
  cards:
    - type: cardThermo
      sleep_timeout: 5
      title: Fancoil Living
      thermo_entity_id: climate.fancoil_living_hvac
```

```    screensaver:
      entities:
        - entity: weather.openweathermap
          color: [255,255,255]
        - entity: sensor.weather_forecast_hourly
          type: 0
          name: ""
          color: [255,255,255]
        - entity: sensor.weather_forecast_hourly
          type: 1
          name: ""
          color: [255,255,255]
        - entity: sensor.weather_forecast_hourly
          type: 2
          name: ""
          color: [255,255,255]
        - entity: sensor.weather_forecast_hourly
          type: 3
          name: ""
          color: [255,255,255]
        - entity: sensor.temperatura_living
          icon: mdi:home-thermometer-outline
          color: [255,255,255]
```

1. In the screensaver section it can only be specified 1 entity for weather but as per docs [here](https://docs.nspanel.pky.eu/stable/config-screensaver/#possible-configuration-values-for-screensaver-config) those can be specified and to enable the alternative layout the sixth entities needs to be specified.. can this option be added?

2. Can locale be enabled for the traslation or does it require much work? The appdaemon version can use `locale: "it_IT"` but in esphome i only found the temp scale that can be customized.

3. I've waited until recently to test because of the missing cardthermo but now that it's in confident to fully switch.. testing i just noticed that the buttons (off, cool, heat) area not centered in the card but are alligned to the left.. the rest seems working perfectly.

Thanks again for this!

olicooper, 2024-08-22:
@r0bb10 I've started working on translations (see the [translations branch](https://github.com/olicooper/esphome-nspanel-lovelace-native/tree/translations)).

It would be great to know if it works for you? It probably needs a more work before merging it to dev.
I imagine that there are a few missing translations - if you spot any missing ones just let me know.. I only have English and German configured so far, but you can create your own custom translation if you copy [en.json](https://github.com/olicooper/esphome-nspanel-lovelace-native/blob/translations/components/nspanel_lovelace/translations/en.json), and update your config to point to the new language file (the path is relative to your esphome config file) e.g.

```yaml
nspanel_lovelace::
  locale:
    language: custom.json
```

Once you've added your new language you can submit a PR on the `translations` branch or I can add for you if you send it me.
A lot of the translations can be found here: https://github.com/joBr99/nspanel-lovelace-ui/tree/main/apps/nspanel-lovelace-ui/luibackend/translations

Chreece, 2024-10-20:
I would like to send the greek translation file:

```
{
  "none": "Κανένα",
  "unknown": "Άγνωστο",
  "preset_mode": "Προκαθορισμένη λειτουργία",
  "swing_mode": "Λειτουργία ταλάντωσης",
  "fan_mode": "Λειτουργία ανεμιστήρα",
  "activity": "Δραστηριότητα",
  "away": "Απουσία",
  "boost": "Ενίσχυση",
  "comfort": "Άνεση",
  "eco": "Οικονομία",
  "home": "Σπίτι",
  "sleep": "Ύπνος",
  "cool": "Δροσιά",
  "cooling": "Ψύξη",
  "dry": "Ξηρό",
  "drying": "Αφύγρανση",
  "fan": "Ανεμιστήρας",
  "heat": "Θερμότητα",
  "heating": "Θέρμανση",
  "heat_cool": "Θέρμανση/Ψύξη",
  "idle": "Αδράνεια",
  "auto": "Αυτόματο",
  "fan_only": "Μόνο ανεμιστήρας",
  "on": "Ενεργοποιημένο",
  "off": "Απενεργοποιημένο",
  "currently": "Τρέχον",
  "state": "Κατάσταση",
  "action": "Ενέργεια",
  "lock": "Κλείδωμα",
  "unlock": "Ξεκλείδωμα",
  "paused": "Παύση",
  "active": "Ενεργό",
  "activate": "Ενεργοποίηση",
  "press": "Πάτημα",
  "run": "Εκτέλεση",
  "speed": "Ταχύτητα",
  "brightness": "Φωτεινότητα",
  "color": "Χρώμα",
  "color_temp": "Θερμοκρασία χρώματος",
  "above_horizon": "Πάνω από τον ορίζοντα",
  "below_horizon": "Κάτω από τον ορίζοντα",
  "position": "Θέση",
  "tilt_position": "Κλίση",
  "start": "Έναρξη",
  "pause": "Παύση",
  "cancel": "Ακύρωση",
  "finish": "Ολοκλήρωση",
  "arm_home": "Οπλισμός σπίτι",
  "arm_away": "Οπλισμός απουσία",
  "arm_night": "Οπλισμός νύχτα",
  "arm_vacation": "Οπλισμός διακοπές",
  "arm_custom_bypass": "Οπλισμός παράκαμψη",
  "disarm": "Αφοπλισμός",
  "disarmed": "Αφοπλισμένο",
  "arming": "Οπλισμός",
  "pending": "Εκκρεμεί",
  "triggered": "Ενεργοποιήθηκε",
  "armed_away": "Οπλισμένο (Απουσία)",
  "armed_home": "Οπλισμένο (Σπίτι)",
  "armed_night": "Οπλισμένο (Νύχτα)",
  "armed_vacation": "Οπλισμένο (Διακοπές)",
  "armed_custom_bypass": "Οπλισμένο (Παράκαμψη)",
  "not_home": "Απουσία",
  "start_cleaning": "Έναρξη καθαρισμού",
  "return_to_base": "Επιστροφή στη βάση",
  "docked": "Στη Βάση",
  "turn_on": "Ενεργοποίηση",
  "turn_off": "Απενεργοποίηση"
}

```

olicooper, 2024-10-20:
Thanks for the translations, I have added them now :)

> also added the days of week and months since there is no more necessary to put them seperately to the .yaml file (days_of_week). I left the values for days in short format also

Unfortunately it doesn't work like that as there are code changes to be made to handle these translations. I have added the month  translations because they were easier to implement but I will have to review the week day translations separately.

olicooper, 2024-10-20:
I decided to move the day_of_week_map to the translations to get it sorted. I did it quickly so testing is minimal. Let me know if it works for you 👍