olicooper/esphome-nspanel-lovelace-native #28: Fails to compile with latest esphome addon

CultusMechanicus, 2024-12-19:
Looks like the type checking in ESPHome 2024.12.1 has got stricter, the compilation process fails with a stack of errors from types.h.

olicooper, 2024-12-20:
This problem is because the [latest esphome](https://esphome.io/changelog/2024.12.0.html#esp-idf) now uses ESP-IDF v5.
This is a bit of a problem because I already tried supporting idf v5 before (which uses [C++ 23](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/cplusplus.html#c-language-standard) - and is the source of these errors) and ran in to similar problems. I wasn't expecting esphome to do a major upgrade like this especially given the issue with platformio not officially supporting idf v5.
I will have a look at what can be done, but I suggest compiling against esphome v2024.11.x for now until I have fixed it.
Thanks for reporting!

CultusMechanicus, 2024-12-20:
Love that the ESPHome Devs don't consider a complete version upgrade to one of the core platforms to be a breaking change...

If anyone needs the old versions of the addon, you can install them from here.

https://github.com/khenderick/esphome-legacy-addons

olicooper, 2024-12-20:
For reference, this was the ugly solution I had before which is going to increase memory usage and uses `reinterpret_cast` which is against the standards... https://github.com/olicooper/esphome-nspanel-lovelace-native/tree/cxx20-support
I might just implement this to get things working again and come back to it when/if I find a better solution.
If anyone has a better idea then feel free to discuss on here 👍

CultusMechanicus, 2024-12-20:
Very dumb question, but can we not just set the `platform:` entry in the yaml back to the previous idf version?

olicooper, 2024-12-20:
I tried updating the platform to `platform: 6.9.0` but it didn't compile because of printf issues in the esphome library.

Anyway, I have applied some updates which you can try out if you'd like? You'll need to use the `cxx23-support` branch for now but I plan to merge it if you say it is working.

I have succesfully compiled against esphome `2024.11.3` (idf v4.4.8) and `2024.12.1` (idf v5.3.2) and uploaded the latest code to my own nspanel which seems to work fine. Memory consumption has increased by ~54KB for my panel but that is to be expected with a new major idf version. I suspect that the code changes have also contributed to the increase in memory consumption but there is no way to know for sure as it wouldn't compile without the changes.

### Example memory consumption difference
```python
## idf v5 - esphome 2024.12.1
RAM:   [=         ]  11.1% (used 36260 bytes from 327680 bytes) + 3252
Flash: [======    ]  64.2% (used 1178377 bytes from 1835008 bytes) + 53852
## platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07

## idf v4 - esphome 2024.11.3
RAM:   [=         ]  10.1% (used 33008 bytes from 327680 bytes)
Flash: [======    ]  61.3% (used 1124525 bytes from 1835008 bytes)
## platform: platformio/espressif32@5.4.0
```

muhlman, 2024-12-21:
I tried the `cxx23-support` branch and so far so good here.

Chreece, 2024-12-22:
I'm also using the branch on one panel and can confirm is working without problems

commit 1178ca92 (2024-12-23): esphome 2024.12 / ESP-IDF v5 support (fixes #28)

* Uses char8_t for icon strings when available due to stricter type checking in c++23.
* Includes icon memory management improvements by reusing existing icon strings when possible.
* Had to be explicit when declaring icon maps due to 'brace initializer list' error.
* Memory consumption is 54kb greater now due to idf upgrade and icon code changes.
closed 2024-12-23 by olicooper