olicooper/esphome-nspanel-lovelace-native #30: [Question] Integrated Climate entity for cardThermo

MarcoBertani93, 2025-01-02:
I there a way to declare a thermostat entity in esphome to be used in a "cardThermo" card directly without the need of homeassistant?

I managed to declare a thermostat entity in esphome, that in recognised in homeassistant, and then use it in the "cardThermo" card. But when the communication between the nspanel and homeassistant is interrupted (for any reason), obviously it stops working.

I even tried to edit "\_\_init__.py" to remove the constraint of passing _thermo_entity_id_ in the format "climate._entity_id_" but no luck:

```
...
nspanel_lovelace:
  id: nspanel
  cards:
  ...
    - type: cardThermo
      title: Thermostat
      thermo_entity_id: thermostat_embedded

  ...

climate:
  - platform: thermostat
    id: thermostat_embedded
```

I tried to take a look at the components files, but I still can't get my head around it. I could try and make some changes to make something work, but I thought it would be better to ask first XD

Great job anyways, I still think this is the best version of the NSPanel custom firmwares

Thank you

olicooper, 2025-01-06:
If I understand you correctly, you want to talk directly with your thermostat without using Home Assistant? The nspanel (this project and all others I know of) are closely connected with HA (or other home automation software) so this is not possible to achieve.

I assume your were trying to create a thermo entity using esphome components like [this one](https://esphome.io/components/climate/thermostat)? That would require soldering relays etc. to the nspanel gpio pins and lots of code changes to hack something together so it is probably not feasible. esphome components describe things that are directly controlled by the device (the nspanel), whereas HA entities are external components that you are making the nspanel aware of so It can control them through HA if that makes sense?

If you can, I would suggest making your HA connection more reliable, for example I have a small pc that is only running HA so I can mess with the rest of my network without breaking HA.

MarcoBertani93, 2025-01-06:
What I am trying to achieve is using the embedded temperature sensor and relays on the nspanel to realise a stand-alone thermostat which can work independently from HA. 

As per your advanced example I created a temperature sensor, and then a climate component based on it. It works as expected when the cardThermo's _thermo_entity_id_ property is the homeassistant entity of the esphome thermostat component. Here is the code I have now:

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.233
    gateway: 192.168.1.254
    subnet: 255.255.255.0
    dns1: 1.1.1.1
    dns2: 8.8.8.8
    
api:
  encryption:
    key: !secret api_encryption
  on_client_connected:
    - logger.log:
        format: "Client %s connected to API with IP %s"
        args: ["client_info.c_str()", "client_address.c_str()"]
  on_client_disconnected:
    - logger.log: "API client disconnected!"
  services:
    ## Service to update the TFT firmware using a URL that is accessible to the nspanel
    - service: upload_tft
      variables:
        url: string
      then:
        - lambda: 'id(nspanel).upload_tft(url);'

ota:
  platform: esphome
  password: !secret wifi_password

esphome:
  name: "nspanel-cameretta"
  friendly_name: "NSPanel Cameretta"
  name_add_mac_suffix: false #true
  comment: "ESPHome Basic NSPanel Example"
  build_path: ./.build/nspanel-basic

esp32:
  board: esp32dev
  framework:
    # ESP-IDF is requried so we can access the NSPanel PSRAM
    # which is on non-standard pins. ESP-IDF also uses less flash than Arduino!
    type: esp-idf

external_components:
  - source:
      type: git
      url: https://github.com/olicooper/esphome-nspanel-lovelace-native
      ref: dev
    refresh: 0s
    components: [nspanel_lovelace]

nspanel_lovelace:
  id: nspanel
  screensaver:
    time_id: homeassistant_time
  cards:
    - type: cardGrid
      id: front_room
      title: Front Room
      entities:
        - entity_id: light.front_room_all
          name: All
        - entity_id: light.front_room_inner
          name: Inner
        - entity_id: light.front_room_outer
          name: Outer
        - entity_id: scene.front_room_all_default
          name: Default
    - type: cardThermo
      sleep_timeout: 60
      title: Thermo Ecobee
      thermo_entity_id: climate.nspanel_cameretta_thermostat_climate_controller
      #thermo_entity_id: thermostat_embedded
      # temperature_unit: fahrenheit
        
logger:
  baud_rate: 115200

## UART for the Nextion display
uart:
  id: uart_nextion
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200

time:
  - platform: homeassistant
    id: homeassistant_time
    timezone: Europe/London

switch:
  ## Physical relay 1
  - platform: gpio
    name: Relay 1
    id: relay_1
    pin:
      number: 19
    restore_mode: ALWAYS_OFF
  ## Physical relay 2
  - platform: gpio
    name: Relay 2
    id: relay_2
    pin:
      number: 22
    restore_mode: ALWAYS_OFF
  ## Turn screen power on/off
  - platform: gpio
    name: Screen Power
    id: screen_power
    entity_category: config
    pin:
      number: 4
      inverted: true
    restore_mode: ALWAYS_ON

binary_sensor:
  - platform: gpio
    name: Left Button
    pin:
      number: GPIO14
      inverted: true
    id: left_button
    #internal: true
    on_click:
    - homeassistant.action:
        action: script.close_cover_cameretta
  - platform: gpio
    name: Right Button
    pin:
      number: GPIO27
      inverted: true
    id: right_button
    #internal: true
    on_click:
    - homeassistant.action:
        action: script.open_cover_cameretta

sensor:
  ## WiFi signal strength sensor
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 300s
  ## Internal temperature sensor, adc value
  - platform: adc
    id: ntc_source
    pin: 38
    update_interval: 60s
    attenuation: 12db
  ## Internal temperature sensor, adc reading converted to resistance (calculation)
  - platform: resistance
    id: resistance_sensor
    sensor: ntc_source
    configuration: DOWNSTREAM
    resistor: 11.2kOhm
  ## Internal temperature sensor, resistance to temperature (calculation)
  - platform: ntc
    id: temperature_sensor
    sensor: resistance_sensor
    calibration:
      b_constant: 3950
      reference_temperature: 25°C
      reference_resistance: 10kOhm
    name: Temperature

# Example dual-point configuration entry
climate:
  - platform: thermostat
    id: thermostat_embedded
    name: "Thermostat Climate Controller"
    sensor: temperature_sensor
    min_cooling_off_time: 300s
    min_cooling_run_time: 300s
    min_heating_off_time: 300s
    min_heating_run_time: 300s
    min_idle_time: 30s
    heat_action:
      - switch.turn_on: relay_1
    cool_action:
      - switch.turn_on: relay_2
    idle_action:
      - switch.turn_off: relay_1
      - switch.turn_off: relay_2
    default_preset: Home
    preset:
      - name: Home
        default_target_temperature_low: 20 °C
        default_target_temperature_high: 22 °C
```

I have a dedicated pc for HA and the connection with my the server is great, I just like my stuff to work even without wifi.

As far as I understand, to obtain what I want, it is necessary to make a "custom" _cardThermo_ card that supports esphome climate components, am I correct?

olicooper, 2025-01-11:
That is an interesting use case! Yes you are correct, you'll need to create a custom thermo card and update the `render` function to read your internal thermo entity. It isn't going to be easy though as the current card is tightly coupled with thermo entities. You'll need to know things like the current themo state, min_temp, max_temp, target_temp_step, hvac_mode etc. but you could also use dummy values too if they are not required. The format of the string in the `render` function is very important, and is described in the upstream repo [HMI readme](https://github.com/joBr99/nspanel-lovelace-ui/blob/main/HMI/README.md). It is likely that the python [build script](https://github.com/olicooper/esphome-nspanel-lovelace-native/blob/dev/components/nspanel_lovelace/__init__.py) will need a bit of modifying to support that use-case (including making the nspanel_lovelace thermo card aware of where to find your internal thermo entity), but it is possible.
There are lots of ways to achieve this but I don't think it'll be something that is added to the repo at this time because I am trying to reach feature-parity with the upstream repo for now and this seems to be a bit of an edge-case (albeit an interesting one!)..
Before you start make sure to plan how you'll implement it and research how things work as it will save you re-writing things later on. Also remember to keep an eye on dynamic memory usage as you are building it. Good luck!