olicooper/esphome-nspanel-lovelace-native #31: [FeatureRequest] Display auto off

andreas-bulling, 2025-01-06:
It would be nice to have a configurable display auto off functionality, i.e. a setting (ideally changeable through HA) that defines after how many seconds the display turns off.

andreas-bulling, 2025-01-12:
Thanks! This allows me to dim the display but will it also come back on when I touch it?

olicooper, 2025-01-12:
Yeah, so the screen (not my code) is the one responsible for the brightness and when it switches the brightness between `active_level` and `inactive_level`.

`inactive_level` is the brightness when the display reaches the timeout configured for the card (which you set with the `sleep_timeout` option in your config). When you touch the display, the screen sets the brightness to the level you set for `active_brightness` and starts a timer which expires at `sleep_timeout`. If you want your screen to stay at the same level all the time you can set the values to be the same/similar i.e. `set_display_dim(100,100)` for 100% brightness or `set_display_dim(0,0)` for permanently off (the screen will never turn on though!). Because the screen controls the brightness, the `set_display_dim` function is limited in what it can do.

closed 2025-03-17 by olicooper