olicooper/esphome-nspanel-lovelace-native #36: Weather forecast not loading after reboot

mmorev, 2025-03-22:
Hello! After power cycling, weather forecast could not be loaded until templated sensor update occurs.

So it looks like this:

[image]

Here is the panel log after boot or on return to main screen (by timeout):
```
[16:51:28][D][nspanel_lovelace:1265]: TFT CMD OUT: weatherUpdate~~~~38066~~3.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C
```

Forecast appears immediately if I force update HA templates from Develooper menu:

```
53:00][D][nspanel_lovelace:1265]: TFT CMD OUT: time~16:53
[16:53:03][D][nspanel_lovelace:1265]: TFT CMD OUT: weatherUpdate~~~~38066~~3.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C
[16:53:03][D][nspanel_lovelace:1265]: TFT CMD OUT: weatherUpdate~~~~38066~~3.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C
[16:53:03][D][nspanel_lovelace:1265]: TFT CMD OUT: weatherUpdate~~~~38066~~3.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C
[16:53:03][D][nspanel_lovelace:1265]: TFT CMD OUT: weatherUpdate~~~~38066~~3.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C~~~~63878~~0.0°C
# Immediately after templates update:
[16:53:03][D][nspanel_lovelace:1265]: TFT CMD OUT: weatherUpdate~~~~38066~~3.0°C~~~~65504~09:00~2.0°C~~~~65504~12:00~1.0°C~~~~38066~15:00~0.0°C~~~~31728~18:00~0.0°C
```

Here is my panel config:
```
...
nspanel_lovelace:
  id: nspanel
  screensaver:
    time_id: homeassistant_time
    weather:
      entity_id: !secret weather_entity_id
...
```

And HA forecast template config:
```
- trigger:
    - platform: time_pattern
      minutes: /15
    - platform: homeassistant
      event: start
    - platform: event
      event_type: event_template_reloaded
  action:
    - service: weather.get_forecasts
      data:
        type: hourly
      target:
        entity_id: weather.gismeteo
      response_variable: hourly
  sensor:
    - name: Weather Forecast Hourly
      unique_id: weather_forecast_hourly
      state: "{{ states('weather.gismeteo') }}"
      attributes:
        temperature: "{{ state_attr('weather.gismeteo', 'temperature') }}"
        temperature_unit: "{{ state_attr('weather.gismeteo', 'temperature_unit') }}"
        forecast: >
                [{%- for n in hourly['weather.gismeteo'].forecast  %}
                   {
                   {%- for k, v in n.items() %}
                     {%- if v is datetime %}
                     "{{ k }}": "{{ v.isoformat() }}",
                     {%- elif v is string %}
                     "{{ k }}": "{{ v }}",
                     {%- else %}
                     "{{ k }}": {{ v }},
                     {%- endif %}
                   {%- endfor %}
                   },
                 {%- endfor %}]

```

Did I misconfigured something?

olicooper, 2025-04-30:
Nothing stands out to me initially. The nspanel subscribes to updates from HA, so when it connects it should get the latest state automatically. It looks like it is not getting the initial state but I don't know why. You might be able to see more info if you connect the nspanel with a serial cable so you can see the logs when it is connecting to wifi and subscribing to entity updates which might shed some light on what is happening?
You could also try updating your config to make it similar to the sample config to see if something is misconfigured?
Is the entity_id in your nspanel config set to `sensor.weather_forecast_hourly` to match your HA template config?

This won't fix anything but another small thing you could add to your HA template config is limiting how many forecast items are included to reduce nspanel memory consumption.
```python
# From this
[{%- for n in hourly['weather.gismeteo'].forecast  %}

# To something like this
[{%- for n in hourly['weather.gismeteo'].forecast[0:5]  %}```

Chreece, 2025-07-19:
Did this brake something?:
https://github.com/joBr99/nspanel-lovelace-ui/releases/tag/v4.7.2

I can only see current temperature and not forecast

olicooper, 2025-07-20:
@Chreece No, I don't use any code from that repository besides the [HMI display code](https://github.com/joBr99/nspanel-lovelace-ui/tree/main/HMI) which was last updated 8 months ago, so I suspect that there might have been a Home Assistant or esphome update that broke it or the forecast data is wrong? Have you updated either of those recently? Can you check what your HA weather entity state is - does it contain the forecast data still?

olicooper, 2025-07-23:
@Chreece I've updated my HA version to `2025.7.3` but it is still working for me.. Could you try the new template and check the entity state looks similar to this:
```yaml
temperature: 20
temperature_unit: °C
forecast:
  - datetime: "2025-07-24T00:00:00"
    condition: sunny
    temperature: 22
  - datetime: "2025-07-25T00:00:00"
    condition: cloudy
    temperature: 16
  - datetime: "2025-07-26T00:00:00"
    condition: cloudy
    temperature: 12
  - datetime: "2025-07-27T00:00:00"
    condition: sunny
    temperature: 14
  - datetime: "2025-07-28T00:00:00"
    condition: sunny
    temperature: 15
friendly_name: Weather Forecast Daily
```

Chreece, 2025-07-24:
Because I'm using this sensor also for other purposes I'm not happy to remove keys.
I will create a new one to remove the null key values.
If you want to test the reasons why it doesn't parse keys with null value you could paste the values from my sensor from above in your developer tools in HA.

closed 2025-07-26 by olicooper