olicooper/esphome-nspanel-lovelace-native #38: discussion: esp-idf options

kiddyfurby, 2025-04-04:
NOT AN ISSUE, but seeing discussion is not enabled on this repo, allow me...

I'd really love to have bluetooth_proxy working (so that my nspanels around the home can track my devices via [bermuda](https://github.com/agittins/bermuda)

But simply enabling bluetooth_proxy resulted in an unresponsive nspanel (luckily OTA is still available)
Seeing that they got bluetooth -Proxy working, I copied over the sdk option from [NSPanel_HA_Blueprint](https://github.com/Blackymas/NSPanel_HA_Blueprint/blob/0c0be2dd26ee12de5d30628c60940976b9ce40f8/components/nspanel_ha_blueprint/__init__.py#L33)
And I am happy to report we can get bluetooth_proxy working onesphome-nspanel-lovelace-native as of esphome 2025.3

Here's the yaml:
```
esp32:
  board: esp32dev
  flash_size: 4MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_D0WD_PSRAM_CLK_IO: "5"
      CONFIG_D0WD_PSRAM_CS_IO: "18"
      CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST: "y"
      CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY: "y"
      CONFIG_ESP32_REV_MIN_3: "y"
      CONFIG_MBEDTLS_DYNAMIC_BUFFER: "y"
      CONFIG_MBEDTLS_DYNAMIC_FREE_CA_CERT: "y"
      CONFIG_MBEDTLS_DYNAMIC_FREE_CONFIG_DATA: "y"
      CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC: "y"
      CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY: "y"
      CONFIG_SPIRAM_RODATA: "y"
      CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP: "y"
```

I tried having only CONFIG_D0WD_PSRAM_CLK_IO / CONFIG_D0WD_PSRAM_CS_IO without the other options and it won't work.
I also remember earlier in this project we used to have those CONFIG_D0WD_PSRAM_CLK_IO lines but they are now removed from the sample config.

Chreece, 2025-04-05:
That was the only component that I was missing! Thanks a lot!

olicooper, 2025-04-30:
Thanks for your suggestion! I suspect that the panel was unresponsive due to running out of memory (it crashes a few times then goes in to safe-mode). The Bluetooth driver uses a lot of memory mostly because of the hard-coded strings I think.
The PSRAM options have been moved to the build script so users don't need it in their config anymore.

Would you be willing to narrow down which options are required to make it work? I could then add a section to the readme for people who want to enable it?
I don't think all of those options will be required but it looks like some of them are shifting runtime data to the PSRAM to free up internal RAM. I would start with only including the following:
```
      CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST: "y"
      CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY: "y"
```
If that doesn't work, try adding `CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY`, then `CONFIG_SPIRAM_RODATA`, then some of the other ones.

I am hoping that NSPanels have the latest ESP chip revision so `CONFIG_ESP32_REV_MIN_3` won't be required? Otherwise that would suggest that there is a bug in earlier chip revisions that make bluetooth proxy not work so some panels will never be able to use the proxy functionality.

The rest of the options are about shifting more data to PSRAM which isn't ideal for performance which is why I am reluctant to include them if they are not required e.g. [CONFIG_SPIRAM_RODATA](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/external-ram.html#move-read-only-data-in-flash-to-psram).

kiddyfurby, 2025-05-06:
This worked.
```
      CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST: "y"
      CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY: "y"
```
And removing CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY resulted in the nspanel no longer flashing, _even via UART._